renesas/fsp #2: Low power analog comparator FSP function R_ACMPLP_Open() overwrites registers on RA2A1

JAALFE, 2020-04-17:
When I use ACMPLP0 and ACMPLP1 and initialize both instances with `R_ACMPLP_Open()` the initialisation of second instance overwrites register for the first instance.
I use IAR EWBARM 8.5 with Renesas Smart Configurator plugin, but it can also reproduce with e² studio.

**To Reproduce**
My settings are for channel 0 and channe 1 are quite the same. Only for channel 1 the instance has the name ACOMP_VO2 and the selected channel is 1.
[image]

As input for channel 1 the values for "Analog Input Voltage Source" and "Reference Voltage Input Source" are the same as for channel 0. This is a bug in Renesas Smart Configurator. I will open another issue for that.

1. Create stack for ACMPLP0 with Renesas Smart Configurator
2. Create stack for ACMPLP1 with Renesas Smart Configurator
3. Open first instance of ACMPLP with `R_ACMPLP_Open(&ACOMP_VO1, &ACOMP_VO1_cfg);`
4. Control registerset. Should be like this:
[image]
5. Open second instance of ACMPLP with `R_ACMPLP_Open(&ACOMP_VO2, &ACOMP_VO2_cfg);`
6. Control registerset. In my opinion the registers for channel 0 shouldn't been touched, but: 
[image]
7. Now the comparators won't work, because the whole input selection is wrong.

**Expected behavior**
I thought, if I use both I can initialize the channels independent to each other. I created two stacks. One for channel 0 and one for channel 1. 
If I set all needed registers directly over CMSIS, both comparators work fine.

I did an workaround to eliminate this problem by setting the lost register values of channel 0 after initialisation of channel 1:
```
R_ACMPLP->COMPSEL0_b.IVCMP0 = 0x01;
R_ACMPLP->COMPSEL0_b.IVCMP1 = 0x01;
R_ACMPLP->COMPSEL1_b.IVREF0 = 0x01;                                         
R_ACMPLP->COMPSEL1_b.IVREF1 = 0x01;                                         
R_ACMPLP->COMPSEL1_b.C1VRF2 = 1;

```
**Desktop:**
 - OS: Windows 10 1903
 - IAR Embedded Workbench for ARM 8.50.1
 - Evaluation board EK-RA2A1

JAALFE, 2020-04-17:
For Renesas Smart Configurator bug see #3

warren-renesas, 2020-04-19:
Hi JAALFE-
We are supporting FSP and RA related issues using our support forum located here:
Support: https://renesasrulz.com/ra/f/forum 

I will copy your issues to the forum and then will add the link in a comment below. You can then go to the forum to see the response.

http://renesasrulz.com/ra/f/forum/16108/low-power-analog-comparator-fsp-function-r_acmplp_open-overwrites-registers-on-ra2a1-2

kamalk-renesas, 2020-07-17:
This issue is moved to support forum, hence closing it from GitHub.

closed 2020-07-17 by kamalk-renesas