renesas/fsp #3: Renesas Smart Configurator generates invalid data for ACMPLP1

JAALFE, 2020-04-17:
**Describe the bug**
Using the Renesas Smart Configurator for RA2A1 and creating a stack for low power analog comparator 1 result invalid data in config structure. 
If you select:
- CMPIN1 (channel 1 only) for Analog Input Voltage Source
- CMPREF1 (channel 1 only) for Reference Voltage Input Source

the generator selects a value from enum, which is not allowed for RA2A1 controller.
Also in initialisation function of FSP `R_ACMPLP_Open()` the value will be set wrong, because the driver will shift the value by 4.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Renesas Smart Configurator
2. Create a new stack for low power analog comparator
3. Use following values:
[image]

4. Press create
5. Check hal_data.c
[image]

[image]

**Expected behavior**
I think the value should be 1 and not 4, because the allowed register values of ACMPLP.COMP0SEL and ACMPLP.COMP1SEL are:
- 0b00
- 0b01
- 0b10

Also the value for channel 1 is shifted by 4 in initialisation routine.

**Desktop**
 - Windows 10 19093
 - RA Smart Configurator Version: 7.8.0.R20200321-2326
 - IAR Embedded Workbench for Arm 8.50.1

warren-renesas, 2020-04-19:
Hi JAALFE-
We are supporting FSP and RA related issues using our support forum located here:
Support: https://renesasrulz.com/ra/f/forum

I will copy your issues to the forum and then will add the link in a comment below. You can then go to the forum to see the response.
http://renesasrulz.com/ra/f/forum/16109/renesas-smart-configurator-generates-invalid-data-for-acmplp1-3

kamalk-renesas, 2020-07-17:
This issue is moved to support forum, hence closing it from GitHub.

closed 2020-07-17 by kamalk-renesas