renesas/fsp #51: RS2L access toGPIO RR/SR only 16bit

alexrayne, 2021-03-15:
Hallow!
i`ve tryed work with gpio.PCNTR3 as 32bit value - set and clear multiple pins at one time, some like  eqivalents to write to port, but only selected bits. 
And stucked that 32bit write to PCNTR3 does nothing. 
```
    port->PCNTR3 = pins & mask; // this is not works
```
only 16bit access to PORR,POSR works.
```
    // this works well
    port->PORR   = ~data & bits_mask;
    port->POSR   = data & bits_mask;
```

what i do wrong?

jeremy-baker, 2021-03-15:
I presume you are using the RA2L1? Which version of the FSP?

This code works for me on the EK-RA2L1 using FSP 2.3.0 to set and clear GPIO P5_01 :- 

    R_PMISC->PWPR = 0x00;
    R_PMISC->PWPR = 0x40;

    R_PFS->PORT[5].PIN[01].PmnPFS_b.PMR = 0; //GPIO

    R_PFS->PORT[5].PIN[01].PmnPFS_b.PDR = 1; //Set Pin as output

    R_PFS->PORT[5].PIN[01].PmnPFS_b.PODR = 1; //P5_01 HIgh using PFS

    R_PFS->PORT[5].PIN[01].PmnPFS_b.PODR = 0; //P5_01 Low using PFS

    R_PORT5->PCNTR3 = 0x00 | (1 << 1); //P5_01 Set using PCNTR3

    R_PORT5->PCNTR3 = 0x00 | ((1 << 1) << 16); //P5_01 Reset PCNTR3

    R_PORT5->POSR   = 0x00 | (1 << 1); //P5_01 Set using PCNTR3

    R_PORT5->PORR   = 0x00 | (1 << 1); //P5_01 Reset PCNTR3

alexrayne, 2021-03-15:
> ```
> R_PORT5->PCNTR3 = 0x00 | (1 << 1); //P5_01 Set using PCNTR3
> 
> R_PORT5->PCNTR3 = 0x00 | ((1 << 1) << 16); //P5_01 Reset PCNTR3
> ```
i guess that this lines are used 16bit access too. do you check disassembler?

alexrayne, 2021-03-15:
> This code works for me on the EK-RA2L1 using FSP 2.3.0 to set and clear GPIO P5_01 :
32bit access not works even from IO-access view at debugger session. But POSR/PORR works well

jeremy-baker, 2021-03-15:
The write to the PCNTR3 register is a 32 bit access (STR instruction) :- 

[image]

The access to the POSR/PORR is a 16 bit access (STRH instruction) :- 

[image]

closed 2021-03-23 by renesas-brandon-hussey