renesas/fsp #58: RA2L SCI not allow FlowControl disabled pins

alexrayne, 2021-04-15:
Hallow!
have  project  SCI interfaces : 
- rs485 *2 
- uart FullDuplex *2

enable FlowControl support -> demands enbling Pin Control, pins on all SCI entityes. 

r_sci_uart.c provide ability to ignore FlowControl on pins that `(p_ctrl->flow_pin != SCI_UART_INVALID_16BIT_PARAM)`

but FSP Configurator not allow leave PinControl and RTS pins Disabled, therefore i can't leave my fullduplex uarts without RTS.

imho, configurator should allow configure p_ctrl->flow_pin as SCI_UART_INVALID_16BIT_PARAM when common FlowControl enabled.

What i do wrong?

jeremy-baker, 2021-04-20:
This appears to be an issue with the configurator in FSP 2.3.0, in that it shows an error when you configure some UART channels to use flow control, and some channels not to use flow control (even though the actual code for the UART driver does appear to support this). As a workaround, you could set a configuration for the UART channel that does not use flow control in the configurator, so that the CTS/RTS selection is RTS (but don't actually configure an RTS pin for that UART channel), and set Pin Control to "Disabled", this then does apeear to generate the correct config structure, the generated code  :-  

  .ctsrts_en = SCI_UART_CTSRTS_RTS_OUTPUT,
#if 0
                .flow_control_pin       = BSP_IO_PORT_00_PIN_00,
#else
  .flow_control_pin = (bsp_io_port_pin_t) (0xFFFFU),
#endif

as SCI_UART_INVALID_16BIT_PARAM is defined as :- 

#define SCI_UART_INVALID_16BIT_PARAM            (0xFFFFU)

This will show an error in the configurator for this channel, however.

Or manually create a configuration  structure for the channels that do not use flow control (so you can set p_extend->flow_control_pin = SCI_UART_INVALID_16BIT_PARAM in the config structure).

I will raise this as an issue with the FSP development team.

jeremy-baker, 2021-04-20:
I have just double checked, and in FSP 2.4.0 this issue has been fixed. The Configurator does not show any errors if some channels use flow control, and some channels don't.

closed 2021-04-26 by renesas-brandon-hussey