renesas/fsp #69: please unstatic SCI callback invoker

alexrayne, 2021-04-20:
hallow!
my application need invoke callbacks assigned to SCI_UART entityes, therefore i have to reimplement invoker function :
static void r_sci_uart_call_callback (sci_uart_instance_ctrl_t * p_ctrl, uint32_t data, uart_event_t event)

please, unstatic (remove static specifier) that invoker function at least. And if you publich in in a r_sci_uart.h header - it would be great.

jeremy-baker, 2021-04-21:
the function  r_sci_uart_call_callback() is not part of the defined public API for the UART interface :- 

https://renesas.github.io/fsp/group___u_a_r_t___a_p_i.html 

this is why it is define as static.

A user can edit the FSP files under the ra/ directory, and the configurator will not overwrite the edited files (unlike the Synergy configurator). You just have to remember to include the edited files if you export the project, or store the project in a version control system, and if you update the version of the FSP make sure you migrate to the latest version of the edited files.

alexrayne, 2021-04-21:
>the function r_sci_uart_call_callback() is not part of the defined public API for the UART interface
unstatic it in r_sci_uart.h - not affects API, nor break any. So that reason is weak, imho. 
Another penalty of static - debugger cant mange it, stack-trace not give readable info.

imho, include this method in api - would be good.

kjassmann-renesas, 2021-04-27:
Our internal functions are static because it allows the compiler to inline them at certain optimization levels. This reduces code size and improves performance.

As @jeremy-baker mentioned, you can edit FSP files. Another alternative is to pass -Dstatic on the command line to make all functions global in scope. Reducing the optimization level should also help with the debugger stack trace.

closed 2021-04-27 by kjassmann-renesas