renesas/fsp #70: R_SCI_UART_BaudSet unexpected change UART mode

alexrayne, 2021-04-22:
Hallow!
just now inspects FSP2.4.0 code of R_SCI_UART_BaudSet and found that it forcibly enables uart RE/RXIE, TE/TXIE (depends on global FSP conf), after baud change.

that strange behaviour, since it easy to keep stat state and recover it properly.

kjassmann-renesas, 2021-08-09:
The register definitions in renesas.h are a superset of all supported MCUs. This is done for scalability. The same r_gpt driver supports all existing RA MCUs. On some MCUs, such as RA6M4, TPCS starts at bit 23.

We understand this is not ideal for users working directly with the register definitions. We are working to improve this in the future.

kjassmann-renesas, 2021-08-09:
Could you please describe your use case? Which desired bit of SCR is cleared unexpectedly?

The r_sci_uart driver always has RE/RIE/TE enabled (depending on global configuration) except during baud rate changes. The only ways SCR could be restored differently are if:
* SCR is modified outside the FSP driver.
* The driver was actively transmitting, and enabled transmit interrupts TIE/TEIE are not restored. This is expected behavior. We do not support resuming a transmission after a baud rate change.

alexrayne, 2021-08-13:
my case is:

* SCR is modified outside the FSP driver.

i used my own code controling UART, and try use FSP routine to setup baud. but luck - this routine useless out of FSP drive scope, reasoned right as you described.

kjassmann-renesas, 2021-08-13:
Ok, thank you for the clarification. I don't see any negative impact to restoring SCR as you suggested. This will be updated in the next release.

closed 2021-08-30 by kjassmann-renesas