renesas/fsp #71: RA2L invalid R_GPT0_GTCR_TPCS_Pos

alexrayne, 2021-04-26:
Hallow!
work with FSP2.4.0 make project for ra2l chip
found that renesas.h have :
#define R_GPT0_GTCR_TPCS_Pos              (23UL)         /*!< TPCS (Bit 23)                                         */
but documentation manual on chip notes that  26:24 TPCS[2:0]  - lay on bit24

kjassmann-renesas, 2021-08-09:
The register definitions in renesas.h are a superset of all supported MCUs. This is done for scalability. The same r_gpt driver supports all existing RA MCUs. On some MCUs, such as RA6M4, TPCS starts at bit 23.

We understand this is not ideal for users working directly with the register definitions. We are working to improve this in the future.

renesas-austin-hansen, 2022-12-05:
This issue has been resolved as of FSP v4.1.0.

closed 2022-12-05 by renesas-austin-hansen