renesas/fsp #83: rm_threadx_port source code is missing from github renesas/fsp repository

hwmaier, 2021-06-10:
This github repo for fsp does contain most source code of the FSP package which is very useful as it allows integration into build systems other than Eclipse/make based e2Studio.

However the source code for the Azure RTOS `rm_threadx` middleware is missing while other Azure RTOS porting middleware like `rm_netx_secure_crypto` and `rm_netxduo_ether` are included.

Could we please have `rm_threadx_port` added to the github FSP?

renesas-brandon-hussey, 2021-06-10:
Thanks for bringing this to our attention. We will fix this. I'm not sure what release this will be in yet. I will update the Milestone when this is scheduled.

hwmaier, 2021-06-10:
@renesas-brandon-hussey Thank you so much. This will make the package more consistent and supports use outside of Eclipse.

renesas-brandon-hussey, 2021-06-16:
@hwmaier this issue will be fixed in the upcoming FSP v3.1.0 release which is scheduled for release by end of June.

hwmaier, 2021-06-17:
Great. I look forward to it.

closed 2021-06-17 by hwmaier