renesas/fsp #85: Problems in configuring sci_uart channel 1

zhengjianxing, 2021-06-11:
in applies-to-v3.0.0, freeRTOS project, i configuring sci_uart channel 1 and  invoking R_SCI_UART_Open (&g_uart1_ctrl, &g_uart1_cfg)，This will slow down the execution of the delay function vTaskDelay(). 
Please accept my English level.
[image]

renesas-brandon-hussey, 2021-06-11:
Hi @zhengjianxing , can you provide information on how you are measuring the slow down? And how did you discover that r_sci_uart was causing this?

zhengjianxing, 2021-06-11:
[image]
[image]

Created a FreeRTOS simple project, only a 100ms led flip task, did not invoking R_SCI_UART_Open (attention:UART channel 1)is normal, after invoking R_SCI_UART_Open , the LED flashing interval will be longer, as long as the timer (GPT, AGT) overflows, the interrupt mode is the same.

zhengjianxing, 2021-06-11:
> Hi @zhengjianxing , can you provide information on how you are measuring the slow down? And how did you discover that r_sci_uart was causing this?

Reply as above

zhengjianxing, 2021-06-12:
Sorry, it should be because the pin is occupied
[image]

closed 2021-06-12 by zhengjianxing