renesas/fsp #86: CMSIS-zone assignment error generating project with P002, P003, or P015 as analog input for RA6T1

mpenate-ellenbytech, 2021-06-16:
Steps to reproduce:
1. Create a new RA project in e2 studio
2. Select MCU: R7FA6T1AD3CFM
3. Select bare metal project
4. Add an ADC stack in FSP configurator
5. Enable ADC0 in pin manager
6. Enable P002, P003 or P015 as analog pins in ADC0 pin manager
7. Attempt to generate in FSP configurator

dlogan01, 2021-06-16:
I can recreate this CMSISS-Zone error with FSP 3.01. The error message also says "Slot name not found P002" (I picked P002 for AN02. I tried it on FSP 2.0.0-beta 1, and I did not get that error. It appears both the 64-pin parts have this problem. The 100-pin parts are fine.

xianghui-renesas, 2021-06-16:
Hi, Marry,

I am planning to get there around 4pm +-15mins.

Would this be ok?

Thanks,
Xianghui

Disclaimer: This message and any files or text attached to it are intended only for the recipients named above and contain information that may be confidential or privileged. If you are not an intended recipient, you must not forward, copy, use or otherwise disclose this communication or the information contained herein. In the event you have received this message in error, please notify the sender immediately by replying to this message, and then delete all copies of it from your system. Thank you.

xianghui-renesas, 2021-06-16:
Hi, Marry,

I am so sorry I won't be able to pick up the package today. Could I pick it up tomorrow morning at around 9:30am?

Thanks,
Xianghui

garyj-renesas, 2021-06-18:
Verified fixed in the 3.1 RC internal release

closed 2021-06-28 by kjassmann-renesas