renesas/fsp #94: ask about #define BSP_CFG_SOFT_RESET_SUPPORTED ((1))

zhengjianxing, 2021-07-23:
MCU:R7FA6M4
#define BSP_CFG_SOFT_RESET_SUPPORTED ((1))
Can this macro definition be configured as 1, the default is 0, what is this for?
[image]

renesas-brandon-hussey, 2021-07-23:
If you are using e2 studio then you can hover over this option to get a little more information.

[image]

The text is `Support for soft reset. If disabled, registers are assumed to be set to their default value during startup.`. This could use more description so I'll look at adding that somewhere. This option is enabling the clock code in the BSP to support when the MCU has a soft reset. With a hard reset all the MCU registers are reset to a certain state. With a soft reset, not all registers are reset. This means that for a soft reset the BSP code cannot rely on registers being in their hard reset state. This requires extra code to check values and make sure all registers are set appropriately. If your application will only be using hard resets then this functionality is not needed and can be removed to reduce the code size of the BSP.

You can find more information about what registers are reset, and when, in the Resets chapter of your MCU's hardware manual.

closed 2021-07-23 by renesas-brandon-hussey

zhengjianxing, 2021-07-26:
> If you are using e2 studio then you can hover over this option to get a little more information.
>
> [image]
>
> The text is `Support for soft reset. If disabled, registers are assumed to be set to their [...]

ok, thanks!