renesas/fsp #111: FreeRTOS functions to get free heap size always return 0

dsabol-ammcoats, 2021-08-30:
FreeRTOS functions xPortGetMinimumEverFreeHeapSize and xPortGetFreeHeapSize always return 0.

<HTML><HEAD>
</HEAD><BODY><DIV class="container"><DIV style="font-size: 10pt;"><TABLE border="0"><TBODY><TR><TD><B>Board: </B> </TD><TD>EK-RA6M1</TD></TR><TR><TD><B>Device:</B> </TD><TD>R7FA6M1AD3CFP</TD></TR><TR><TD><B>Toolchain: </B> </TD><TD>GCC ARM Embedded</TD></TR><TR><TD><B>Toolchain Version: </B> </TD><TD>9.2.1.20191025</TD></TR><TR><TD><B>FSP Version: </B> </TD><TD>3.0.0</TD></TR><TR><TD><B>Project Type: </B> </TD><TD>Flat</TD></TR></TBODY></TABLE></DIV></DIV></BODY></HTML>

renesas-brandon-hussey, 2021-08-30:
Hi @dsabol-ammcoats , I have a couple of questions.

1. What heap implementation are you using?
2. What is your configured heap size?

[image]

I did a quick test using Heap 1 and a 4KB heap and xPortGetFreeHeapSize() returned the expected value.

dsabol-ammcoats, 2021-09-01:
Hi,

I am using heap 4 with a 4KB heap.  I will try making a new project from scratch using the blinky template project and see if I still see the issue.

renesas-brandon-hussey, 2021-09-30:
@dsabol-ammcoats I created a v3.0.0 project with heap 4 as you mentioned. I added a thread and included the code below in the thread entry file.

```c
#include "new_thread0.h"

size_t min_heap;
size_t free_heap;
uint32_t * p_malloced;

/* New Thread entry function */
/* pvParameters contains TaskHandle_t */
void new_thread0_entry(void *pvParameters)
{
    FSP_PARAMETER_NOT_USED (pvParameters);

    /* Heap not initialized yet, so will be 0 */
    min_heap = xPortGetMinimumEverFreeHeapSize();
    free_heap = xPortGetFreeHeapSize();

    while (1)
    {
        p_malloced = pvPortMalloc(4);

        min_heap = xPortGetMinimumEverFreeHeapSize();
        free_heap = xPortGetFreeHeapSize();

        vTaskDelay (1);
    }
}
```

The first call to xPortGetMinimumEverFreeHeapSize() and xPortGetFreeHeapSize() do in fact return 0. The reason for this is that FreeRTOS does not initialize the heap system until you call pvPortMalloc() the first time. This is from heap_4.c.

```c
void * pvPortMalloc( size_t xWantedSize )
{
    BlockLink_t * pxBlock, * pxPreviousBlock, * pxNewBlockLink;
    void * pvReturn = NULL;

    vTaskSuspendAll();
    {
        /* If this is the first call to malloc then the heap will require
         * initialisation to setup the list of free blocks. */
        if( pxEnd == NULL )
        {
            prvHeapInit();
        }
        else
        {
            mtCOVERAGE_TEST_MARKER();
        }
```

Using my sample code above, I could see that min_heap and free_heap were updated with non-zero values each time the while() loop was executed. You don't see this issue with heap 1 because it does not require a heap init call. Below is an example from heap_1.c.

```c
size_t xPortGetFreeHeapSize( void )
{
    return( configADJUSTED_HEAP_SIZE - xNextFreeByte );
}
```

closed 2021-09-30 by renesas-brandon-hussey