renesas/fsp #114: USB PCDC

rmatris, 2021-09-01:
R_USB_DataSizeGet returns datasize of 0xffffffd0 (or variants of large numbers) when sending greater than 1 byte of data at a time.
[image]

This seems to be related to the code below. 
r_usb_pdriver.c
usb_pvnd_read_complete: 
ctrl.data_size = mess->read_req_len - mess->tranlen; << results in a negative value or large unsigned value.

rmatris, 2021-09-01:
The read size is not updated when calling R_USB_Read.
In the example project "USB_PCDC_EK_RA4M2_EP",  case USB_STATUS_CONFIGURED is always hit first which reads out 8 bytes.
When case USB_STATUS_READ_COMPLETE is hit, the readsize is not updated, and is set to the old/previous value of 8.

TakahiroMuranaka, 2021-09-03:
Please do not use the instance variable "_**g_basic0_ctrl**_" when getting the received size.

Please get the received size as follow.

[image]

closed 2021-09-03 by rmatris