renesas/fsp #115: R_BSP_RegisterProtect module not functional before BSP_WARM_START_POST_C stage

alexrayne, 2021-09-02:
Hallow!
try some control on CGC at `R_BSP_WarmStart` before `BSP_WARM_START_POST_C` stage (try `BSP_WARM_START_POST_CLOCK`)
cgc functional hungs, since it can`t unprotect registers access via `R_BSP_RegisterProtectXXX` functions. 

`R_BSP_RegisterProtectXXX` functional rely on `g_protect_counters`- that if initiated at bss data setup. This bss initialisation goes right after  `BSP_WARM_START_POST_CLOCK` stage, and at this thime  RegisterProtect API functional.

IMHO, better declare `g_protect_counters` as unitialised, and cleanup it right before `BSP_WARM_START_RESET` stage. So any protected registers functionality availiable in `R_BSP_WarmStart` in any stages.

alexrayne, 2021-09-08:
same mention relates on bsp_io.c: g_protect_pfswe_counter

closed 2021-12-08 by kjassmann-renesas

kjassmann-renesas, 2021-12-08:
Resolution: This behavior can be enabled with the new `Early BSP Initialization` configuration in v3.5.0.

https://renesas.github.io/fsp/group___b_s_p___m_c_u.html#:~:text=Early%20BSP%20Initialization