renesas/fsp #116: configurator doesn't describe ISR conflicts

alexrayne, 2021-09-06:
Hallow!
configuring interrupts in FSP configurator, regulary got conflicts on adding new ISR. But conflict message not shows - wich isr are conflict with new one. i have to explore in with some non-trivial, or hard way. 
Expecialy it confuses when conflicts with ISR allocated by stack modules.

IMHO, show - wich ISR are in cinflicts, would be great

renesas-brandon-hussey, 2021-09-07:
Hi @alexrayne can you please provide the following:

1. Which MCU you are using
2. Which FSP version you are using
3. Screenshots or or more information on the issue you are encountering. I am having trouble understanding the issue. I think it is related to conflicts in allocating interrupt slots on a RA2 MCU.

alexrayne, 2021-09-08:
Hallow!
1) work with RA2L1
2) any FSP. latest was v3.2
3) you right - this is about irq slots allocating conflicts. and it looks as
[image]
it shows only error mark. no any info - why, wich isr are conflicts.

renesas-austin-hansen, 2024-01-22:
Closing as duplicate of #142

closed 2024-01-22 by renesas-austin-hansen