renesas/fsp #117: FSP v3.2 IIC loose  st_iic_master_extended_cfg.timeout_scl_low field

alexrayne, 2021-09-06:
Hallow!
porting my app to FSP v3.2 got this error:
```
../ra_gen/hal_data.c:404:51: error: 'iic_master_extended_cfg_t {aka const struct st_iic_master_extended_cfg}' has no member named 'timeout_scl_low'; did you mean 'timeout_mode'?
 { .timeout_mode = IIC_MASTER_TIMEOUT_MODE_SHORT, .timeout_scl_low = IIC_MASTER_TIMEOUT_SCL_LOW_ENABLED,
                                                   ^~~~~~~~~~~~~~~
                                                   timeout_mode
../ra_gen/hal_data.c:404:69: error: 'IIC_MASTER_TIMEOUT_SCL_LOW_ENABLED' undeclared here (not in a function); did you mean 'IIC_MASTER_TIMEOUT_MODE_LONG'?
 { .timeout_mode = IIC_MASTER_TIMEOUT_MODE_SHORT, .timeout_scl_low = IIC_MASTER_TIMEOUT_SCL_LOW_ENABLED,
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                     IIC_MASTER_TIMEOUT_MODE_LONG
```

looks that st_iic_master_extended_cfg now not have timeout_scl_low  field, but configurator generates hal_data structrures with it.

aniruddhabajaj-renesas, 2021-09-07:
Hello!

Can you confirm r_iic_master.h is correctly getting generated for you ?
In your e2studio project, you can delete or rename the existing r_iic_master.h under ra/fsp/inc/instances/ and hit the 'Generate Project Content' button.

alexrayne, 2021-09-08:
thats my miss - i have a copy of fsp3.1 driver in a project

closed 2021-09-08 by alexrayne