renesas/fsp #151: Reopening e2 studio removes User Events from Interrupts Configuration

jachimop, 2021-11-17:
I added new event CAN1 COMFRX (Common FIFO receive) for RA6M5 project. After closing e2 and opening again this event is gone.

jachimop, 2021-12-08:
Any suggestions how to solve this issue? This is rather critical for my team, because it forces every developer to manually add the interrupt every time they open the project in e2 and regenerate FSP code.

renesas-brandon-hussey, 2021-12-08:
@jachimop sorry for not getting back to you sooner. Please note that official support for RA MCUs and FSP is on [RenesasRulz](https://renesasrulz.com/ra). The FSP development team tries it best to keep up with GitHub Issues, but we are a separate team from technical support. For that reason we typically need specific information about how to reproduce an issue. If you have a critical issue, or need help with debugging the issue, we recommend you go through the official support channels.

In regards to the issue, we need more information. I did some quick testing and was not able to reproduce this issue. Some information we need:

1. What FSP version are you using?
2. Can you give us steps to reproduce the issue?

jachimop, 2021-12-16:
Ok, the issue can be reproduced even on the new projects. Here are the steps:

1. Open e2 Version: 2021-10 (21.10.0).
2. Create 2 TrustZone projects - secure and non-secure, FSP version 3.4.0 and R7FA6M5BH2CBG device.
3. Add CANFD stack in the FSP Configuration of the non-secure project.
4. Add CAN0 COMFRX interrupt in the Interrupts Configuration (non-secure).
[image]
5. Close e2 and open again. The added ISR should disappear.
[image]

renesas-brandon-hussey, 2021-12-20:
@jachimop thanks for the information. I have reproduced the issue. I will follow up with the e2 studio team.

renesas-brandon-hussey, 2021-12-21:
@jachimop I have confirmed the issue with the e2 studio team. Unfortunately, there is not a known workaround for e2 studio 2021-10 at this time. We have confirmed that the issue is fixed in the next e2 studio release, 2022-01, which is scheduled to be released around the end of January 2022. It will be listed here when available.

https://www.renesas.com/us/en/software-tool/e-studio

The next FSP release will use the e2 studio 2022-01 build. This release, 3.6.0, is currently scheduled for the end of February 2022.

closed 2022-02-25 by JGrenard