renesas/fsp #152: Many warnings occur due to FSP

mertyilmaaz, 2021-11-28:
The cause of the error is capital "İ" letter. The project files created by FSP has some capital "İ" letters and this results in many warnings, if corrected manually, warnings are eliminated.
And this capital "İ" letters are only included in ifndef preprocessor statements.
[image]

mertyilmaaz, 2021-11-29:
The problem is solved, if you face with the problem what you need to do is to change your windows display language to English. There may be alternative solutions, this is how I dealt with it.

renesas-brandon-hussey, 2021-11-29:
@mertyilmaaz what was your display language set to previously?

closed 2024-01-22 by renesas-austin-hansen