renesas/fsp #156: Failed install setup_fsp_v3_4_0_rasc_v2021-10.exe on windows7 x64

Femtometer, 2021-12-03:
Install setup_fsp_v3_4_0_rasc_v2021-10.exe

[image]

Femtometer, 2021-12-03:
Translation: Failed running this program, because this computer missing api-ms-win-core-path-l1-1-0.dll. Reinstall to sovle this problem.

Femtometer, 2021-12-03:
Python 3.9+ does  not support win7, does this software use python?
https://superuser.com/questions/1660386/api-ms-win-core-path-l1-1-0-dll-is-missing-from-your-computer

renesas-brandon-hussey, 2021-12-03:
HI @Femtometer , which operating system are you using? e2 studio 2021-10 requires Windows 8 or newer. This is documented in the release notes. 

https://www.renesas.com/us/en/document/rln/e-studio-2021-10-release-note?language=en&r=488826

Femtometer, 2021-12-11:
@renesas-brandon-hussey I'm using Win7, Solved by copy folder from an installed computer.

closed 2021-12-11 by Femtometer