renesas/fsp #180: Missing SPI 'SSL Signal Level Keeping' configuration

jachimop, 2022-02-01:
I noticed that the SPI Driver does not include a setting for SSL Signal Level Keeping and there is only the BSP_FEATURE_SPI_HAS_SSL_LEVEL_KEEP macro in bsp_feature.h that causes the SPI Driver code to always enable level keeping (R7FA4M2AB3CFP). For now I can modify the generated code manually, but it looks to me that this option should be part of the stack configuration.

jritchey-renesas, 2022-02-03:
Hi @jachimop, In master mode, the SPI Driver sets SSLKP in order to ensure that SSL is not negated after every [byte,halfword,word]  during the transfer. With SSLKP set to 1, the driver will only negate SSL at the end of the transfer.

Could you provide some more information about what you are trying to achieve?

jachimop, 2022-02-10:
It is correct, that is what I need to adjust during runtime and generated SPI Driver (FSP v3.5.0) do not support it in any way. Right now I modify SSLKP directly through register before executing R_SPI_Read(...). It would make more sense to have SSL Level Keeping option as function parameter and BSP_FEATURE_SPI_HAS_SSL_LEVEL_KEEP macro should not exist.

jritchey-renesas, 2022-02-17:
The SSL Level Keeping option will likely not be added as an argument to SPI functions for compatibility reasons (Support could potentially be added in other ways). I'm still not entirely sure I understand what is the use case for asserting and negating SSL for every byte in the transfer. Is this done in order to repeat single-byte transfers back to back?

jachimop, 2022-02-18:
I need to periodically send to my device either one 64-bits frame or sixteen 32-bits frames, one after another. In the first case I use R_SPI_Read with SPI_BIT_WIDTH_32_BITS, length 2 and with enabled level keeping. In second case I have SPI_BIT_WIDTH_32_BITS, length 16 and disabled level keeping.

jritchey-renesas, 2022-02-24:
That is really cool that you were able to leverage one of the more advanced features of the peripheral. Originally we didn't anticipate a lot of customers needing this feature which is why SSLKP is hard-coded in the driver.

Unfortunately we don't currently have any plans for supporting this (We might consider it in a future release).

closed 2022-02-24 by jritchey-renesas