renesas/fsp #182: MCUboot default Image Header Size limits interrupt slots in application

kjassmann-renesas, 2022-02-03:
## Issue

When `Image 1 Header Size (Bytes)` is set to `0x80` (default setting), the number of interrupt slots available to the image application is limited to 16 (slots 0-15 on the Interrupts tab).

## Workaround

In order to have access to all interrupts in the image application, update `Image 1 Header Size (Bytes)` (and  `Image 2 Header Size (Bytes)` if 2 images are used) as follows:

 * Cortex-M23 devices and RA4M1/RA4W1: (32 interrupt slots): Must be a multiple of 0x100
 * Cortex-M33 and Cortex-M4 devices excluding RA4M1/RA4W1 (96 interrupt slots): Must be a multiple of 0x200

CeciliaWongRECH, 2022-02-15:
Must be a multiple of 0x100.
If the image header size is set to 0x300, then the effective value is 0x200. Is it right?
So it's not multiple of 0x100 but (power of 2 * 0x100). That would be more appropriate.

kjassmann-renesas, 2022-02-15:
@CeciliaWongRECH 

Yes, thank you. I updated the description to specify power of 2 multiple.

closed 2022-02-25 by JGrenard