renesas/fsp #212: FSP AppImage Fails on Debian 4.19.98

jsbosch, 2022-07-14:
File: setup_fsp_v3_8_0_e2s_v2022-04.AppImage

Run from UI generates a small square window without content then quickly exits. 

Run from terminal results in the following error messages, which appear to be from trying to open two missing image files (though the installer runs fine on Ubuntu). Before exiting, it also produces a small blank square window.

`jeffb@debian2:~/temp/mate3s/tools$ ./setup_fsp_v3_8_0_e2s_v2022-04.AppImage
(java:10802): GdkPixbuf-CRITICAL **: 12:47:30.521: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
(java:10802): GdkPixbuf-CRITICAL **: 12:47:30.521: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
(java:10802): GdkPixbuf-CRITICAL **: 12:47:30.522: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
(java:10802): GdkPixbuf-CRITICAL **: 12:47:30.522: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
(java:10802): Gtk-CRITICAL **: 12:47:30.522: gtk_window_resize: assertion 'width > 0' failed
(SWT:10802): Gtk-WARNING **: 12:47:33.102: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
`

Run with SUID produces same results. 

Run with sudo produces display server error messages. Note that the log file repeats the messages below.

`jeffb@debian2:~/temp/mate3s/tools$ sudo ./setup_fsp_v3_8_0_e2s_v2022-04.AppImage 
[sudo] password for jeffb: 
Invalid MIT-MAGIC-COOKIE-1 keyUnable to init server: Could not connect: Connection refused
Installer: Cannot open display: 
Invalid MIT-MAGIC-COOKIE-1 keyUnable to init server: Could not connect: Connection refused
Installer: Cannot open display: 
Invalid MIT-MAGIC-COOKIE-1 keyUnable to init server: Could not connect: Connection refused
Invalid MIT-MAGIC-COOKIE-1 keyUnable to init server: Could not connect: Connection refused
Installer: Cannot open display: 
Installer:
An error has occurred. See the log file
/root/.eclipse/1915181882_linux_gtk_x86_64/configuration/1657816258518.log.
`

I've tried pretty much every "fix" I've found online to no avail. Installer runs fine on Ubuntu.

renesas-brandon-hussey, 2022-07-18:
There does not appear to be a Debian 4.19.98. I'm guessing you mean that you're using kernel 4.19.98. Which distro are you using?

jsbosch, 2022-09-26:
Apologies for the late response. 

I worked around this by installing on a (gasp) Windows workstation. I had tried all the fixes I had found at the time but none worked. I was able to install it on a Ubuntu VM but by that time I was already invested in the time on Windoze. Thanks for following up, though.

jsbosch, 2022-09-26:
Here's the output of running with the extract-and-run option. Hope that helps. Note that I had tried image type database fixes recommended online. 

> tmp$ ./setup_fsp_v3_8_0_e2s_v2022-04.AppImage --appimage-extract-and-run

> (java:2191): GdkPixbuf-CRITICAL **: 12:23:18.700: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
> 
> (java:2191): GdkPixbuf-CRITICAL **: 12:23:18.700: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
> 
> (java:2191): GdkPixbuf-CRITICAL **: 12:23:18.700: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
> 
> (java:2191): GdkPixbuf-CRITICAL **: 12:23:18.700: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
> 
> (java:2191): Gtk-CRITICAL **: 12:23:18.700: gtk_window_resize: assertion 'width > 0' failed
> 
> (SWT:2191): Gtk-WARNING **: 12:23:20.965: Could not load a pixbuf from icon theme.
> This may indicate that pixbuf loaders or the mime database could not be found.
> **
> Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
> Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
>

jsbosch, 2022-09-26:
I'm willing to try other things if you think it would be valuable. Otherwise, feel free to close this. Thanks again.