renesas/fsp #228: E2 studio does not generate define needed to OSPI work in DMA mode

GluchowD, 2022-08-12:
Hi,
I am working on a project using e2 studio (3.6, but I also tested it on 3.8) and RA6M5 MCU. I use a OSPI interface for RAM and Flash, When I enable DMA for OSPI Flash and generate configuration. The r_ospi_cfg.h file miss define OSPI_CFG_DMAC_SUPPORT_ENABLE (1). I have to use a DMA, because normal implementation is not robust when for example interrupt occur while writing operation is executed. Can I ask you to provide instructions on forcing the generation of this define?
[image]
[image]
[image]

renesas-austin-hansen, 2022-08-16:
Hi Dominik!  Global defines can be added by navigating to **Project -> Properties** then **C/C++ Build -> Settings**.  There should be an option to change the defines provided in toolchain commands there.  For example, this is what the window looks like for GCC:

[image]

renesas-brandon-hussey, 2022-08-16:
Hi @GluchowD , this is an issue where both OSPI Flash and OSPI RAM are attempting to create the r_ospi_cfg.h file. This is scheduled to be fixed in the next release of FSP.

closed 2022-08-29 by GodavarthyAravinda