renesas/fsp #229: E2 Studio Is Unable to See FSP version 3.8.0

dengelofgatesair, 2022-08-18:
I don't know if this is an FSP issue or if it is an E2 Studio issue. However, we getting the following error:

`FSP version 3.8.0 is not installed. Version 3.6.0 will be selected`

https://renesasrulz.com/tools/e2studio/f/e2studio-forum/19559/newly-installed-fsp-version-not-recognized-after-e2-restart

We don't seem to be the only ones with this issue.

[image]

renesas-brandon-hussey, 2022-08-18:
Hi @dengelofgatesair , this message is telling you that the e2 studio installation you are using does not have the FSP v3.8.0 packs installed. Each e2 studio installation will have it's own collection of FSP versions. In other words, FSP installations are not globally available to all e2 studio installations.

Maybe you have multiple e2 studio installations and 1 has FSP v3.8.0 and the other has FSP v3.6.0? To fix this you will need to install FSP v3.8.0 into the e2 studio installation that you are using for this project. You can install a specific FSP version into an e2 studio installation using the standalone installer which is included with each release. [Here](https://github.com/renesas/fsp/releases/download/v3.8.0/FSP_Packs_v3.8.0.exe) is the self-extracting FSP v3.8.0 standalone installer. The installer has directions for how to complete the install. Please check the e2 studio version required for each FSP version on the main page of this repo under the "Setup Instructions" section. FSP v3.8.0 requires e2 studio 2022-04 or later.

dengelofgatesair, 2022-08-18:
This worked.

closed 2022-08-18 by dengelofgatesair