renesas/fsp #238: CMSIS accuracy

mdurian, 2022-09-09:
I am not using FSP in my project, but I am using the CMSIS definitions. I recently noticed that the TPCS bitfield in the GPT.GTCR register on the RA6M3 does not agree with the documentation (and the documentation seems to match the chip behavior). CMSIS has TPCS as a 4-bit field starting at bit 23, while the documentation has it as a 3-bit field starting at bit 24. (Renesas RA6M3 Group User's Manual, Rev 1.10, R01UH0886EJ011 section 23.2.12)

I reported this to Renesas support and after some exchanges, their conclusion was that FSP expects this field to be 4-bits and the FSP code works correctly, thus not a bug. Furthermore, the CMSIS definitions are "strictly for FSP. There is no warranty on their accuracy outside FSP." Is this a correct characterization? Is CMSIS only for FSP and its accuracy should not be relied on?

If that is not true, then I'm reporting a bug. The R_GPT0->GTCR_b.TPCS bitfield is incorrect in R7FA6M3AH.h.

If it is true that CMSIS is intended only for use by FSP and the TPCS definition intentionally differs from the on-chip reality, is there a list of these discrepancies so I can make adjustments on my end? The thought of tracking them down by hand daunts me.

renesas-brandon-hussey, 2022-09-12:
HI @mdurian .

> CMSIS definitions

Just to be clear for others that might read this ticket, this issue is in regards to the register access header file that is supplied with FSP. Some might also call this file a SFR (special function register) file or if you've used existing Renesas MCUs maybe an iodefine file. "CMSIS" is a broad package of different components. We ship this header file as part of our implementation of CMSIS-Core.

> Furthermore, the CMSIS definitions are "strictly for FSP. There is no warranty on their accuracy outside FSP." Is this a correct characterization? Is CMSIS only for FSP and its accuracy should not be relied on?

A better way of stating this is that the header file is not tested outside of its use in FSP by the FSP team. There could be issues when accessing registers and bitfields that are not used by FSP.

The specific issue of the TPCS bitfield is different since we are using it in FSP. The issue there is that the TPCS bitfield is not the same length on all RA MCUs. On the RA6M3 it is 3-bits as you mention. On the RA6M5 it is 4-bits as you see in the header file. The 4-bit length is used for the RA6M3 currently because the header file used in FSP was the same for all RA MCUs. In the FSP code we can use the 4-bit length and make sure we only set it to valid values. For users not using FSP this can cause confusion.

We are aware of this issue and have been working to fix it. In FSP v4.0.0, we transitioned from having a single register access header file to ones specific to each MCU. You can see the files [here](https://github.com/renesas/fsp/tree/v4.0.0/ra/fsp/src/bsp/cmsis/Device/RENESAS/Include). There is still more work to do though. The specific issue in this ticket is currently slated to be fixed in the next FSP release, FSP v4.1.0.

The person you spoke to in support was correct that in the past we have not had a clear path to fixing issues like these. Starting with FSP v4.0.0, we are making progress and can start addressing these types of issues. In summary, please report these types of issues. I will follow up with support in the coming weeks so they understand this change as well.

mdurian, 2022-09-12:
Hi @renesas-brandon-hussey ,
Thank you for the clarification.

With regards to reporting these kinds of issues, I've got some more related to the R_FACI_HP register block. I've only checked this against the R7FA6M3AH.h header. I don't know if it applies to the other devices, as well.

I'm using the "Flash Memory Programming" application note as a reference (Rev 1.20, R01AN5367EU0120). Perhaps there is a better reference that would resolve some of these issues.

The following registers are represented in R7FA6M3AH.h, but are not documented in the application note (or the RA6M3 User's Manual):

- FMEPROT
- FBPROT0
- FBPROT1
- FPESTAT

The following bitfields are present in the header file, but are not documented in the app note:

- FASTAT.ECRCT
- FAEINT.ECRCTIE

The following register is documented in the application note, but is not present in the header file:

- FCKMHZ

closed 2022-10-24 by GodavarthyAravinda

mdurian, 2022-10-25:
Perhaps it was intentional, but fsp-4.1.0 still does not include a definition for the FCKMHZ register. It contains FCKMHZSA, which is in R_FCACHE_Type, but it is different from the FCKMHZ related to FACI. FCKMHZ is the "Data Flash Access Wait Register".

It is a bit of an odd duck since its address is pretty far from the rest of the FACI registers. It resides at address 0x407FC040, while most of the FACI registers are at 0x407FExxx. I don't see any other registers in the 0x407FCxxx memory area.

Maybe FCKMHZ was intentionally excluded to avoid creating a register block specifically to hold that one register. Anyway, I thought I'd point it out in case it was an oversight.

From the "Flash Memory Programming" application note version 1.20.
[image]