renesas/fsp #243: R_IIC_Master cannot change slave address after NAK when IIC_MASTER_CFG_PARAM_CHECKING_ENABLE is set

ijager, 2022-10-12:
When `IIC_MASTER_CFG_PARAM_CHECKING_ENABLE` is set, the API does not allow to change the slave address via `R_IIC_MASTER_SlaveAddressSet` when the previous transfer failed with a NACK.

In `iic_master_err_master()` the `p_ctrl->restart` flag is not set to false. Therefore `R_IIC_MASTER_SlaveAddressSet` thinks a transfer is still in progress:

```c
    /* Fail if there is already a transfer in progress */
    FSP_ERROR_RETURN(((0 == p_ctrl->remain) && (false == p_ctrl->restart)), FSP_ERR_IN_USE);
```

My workaround is to manually reset the restart flag in the user callback:

```c

void i2c_master_callback(i2c_master_callback_args_t *p_args)
{
    if (NULL != p_args)
    {
        /* capture callback event for validating the i2c transfer event*/
        i2c_event = p_args->event;

		switch (i2c_event) {
			case I2C_MASTER_EVENT_ABORTED:
                            g_i2c_master_ctrl.restart = false;
                }
```

I believe this should be handled inside the `R_IIC_Master` driver.

SeanMeng0509, 2022-12-05:
Hi @ijager, may I know more about the `previous transfer` before R_IIC_MASTER_SlaveAddressSet was called in your project? 
I tried to reproduce the problem by calling `R_IIC_MASTER_Write(&g_i2c_master0_ctrl, write_buffer, 1, false)` , but didn't see this error.  However, if called with the `restart` argument set to true, then the following call to `R_IIC_MASTER_SlaveAddressSet` will fail. The 'restart' argument should be `false` unless it is in the middle of transfers as required by the i2c protocol.

ijager, 2022-12-07:
Here is my full code for context. The `response_len` flag decides whether we only do a write or both a write and a read with a `restart`.

```c

void i2c_write_buffer(uint8_t slave_address, const uint8_t* buffer, size_t buffer_len, size_t response_len)
{
	if (context.transaction_busy) {
		APP_PRINT("i2c still busy, reject\n");
		return;
	}

	context.slave_address = slave_address;
	context.read_length = response_len;
	context.transaction_busy = true;

	// Copy data to local buffer
	for (uint8_t i=0; i < buffer_len; i++) {
		context.buffer[i] = buffer[i];
	}

    fsp_err_t err = R_IIC_MASTER_SlaveAddressSet(&g_i2c_master_ctrl, slave_address, I2C_MASTER_ADDR_MODE_7BIT);
    if (err) {
        APP_PRINT("i2c error changing slave address: %d\n", err);
    }

	// restart means a write operation is followed by a read operation; no need to send address again
	bool should_restart = response_len > 0;

	err = R_IIC_MASTER_Write(&g_i2c_master_ctrl, context.buffer, buffer_len, should_restart);
	if (err) {
		APP_PRINT("i2c error for write: %d\n", err);
	}

}

void i2c_master_callback(i2c_master_callback_args_t *p_args)
{
    if (NULL != p_args)
    {
        /* capture callback event for validating the i2c transfer event*/
        context.i2c_event = p_args->event;

		switch (context.i2c_event) {
			case I2C_MASTER_EVENT_TX_COMPLETE:
				if (context.read_length > 0) {
					R_IIC_MASTER_Read(&g_i2c_master_ctrl, context.buffer, context.read_length, false);
				} else {
					context.transaction_busy = false;
				}

			break;
			case I2C_MASTER_EVENT_RX_COMPLETE:
				context.transaction_busy = false;
				if (context.recv_callback) {
					for (uint8_t i=0; i < context.read_length; i++) {
						context.recv_callback(context.slave_address, context.buffer[i]);
					}
				}

			break;
			case I2C_MASTER_EVENT_ABORTED:
            	context.transaction_busy = false;
                g_i2c_master_ctrl.restart = false; // make sure we can change the i2c address after an abort
				APP_PRINT("i2c aborted for slave address 0x%02X\n", context.slave_address);
			break;
		}
    }
}
```

SeanMeng0509, 2022-12-07:
Hi @ijager, thank you very much for the information. The restart flag is for the IIC master to hold the bus after an I2C transaction, so there is no problem with your code for the normal happy case. I will try to reproduce the problem and make a fix.

ijager, 2022-12-14:
> Also FSP has a workaround for recovering from error cases: R_IIC_MASTER_Abort(), which will safely abort any in-progress transfer and force the IIC peripheral into ready state.

Actually I think I have found a case where `R_IIC_MASTER_Abort()` does not work.

Abort has no effect if these checks in `iic_master_abort_seq_master()` are not met:

```c
    if ((0U != p_ctrl->remain) || (p_ctrl->restarted) || (true == p_ctrl->err))
```

However in `iic_master_run_hw_master()` we can have the case that there was a timeout waiting for the peripheral busy bit (`BBSY`). This results in a FSP_ERROR_RETURN, but it does not actually set the `p_ctrl->err` flag. So it can never recover from this situation because Abort will not do anything.

I fixed it by adding the following to `iic_master_run_hw_master()` just before the `FSP_ERROR_RETURN((0U != timeout_count), FSP_ERR_IN_USE);`.

```c
if (timeout_count == 0) {
    p_ctrl->err = true;
}
```
Here is the full function:

```c
static fsp_err_t iic_master_run_hw_master (iic_master_instance_ctrl_t * const p_ctrl)
{
    /* IICn operates using PCLKB. The ratio ICLK (System Clock)/PCLKB gives the CPU cycles for 1 ICBRL count.
     * (ICLK/PCLKB)*ICBRL gives the CPU cycles for entire ICBRL count.
     * Since each time we loop the timeout count will be decremented by 1 this would require at least 4 CPU clocks,
     * making the final timeout count as:
     * Timeout = ((ICLK/PCLKB)*ICBRL)/4.
     */
    volatile uint32_t sysdiccr =
        (uint32_t) (R_SYSTEM->SCKDIVCR & (IIC_MASTER_SYSDIVCK_ICLK_MASK | IIC_MASTER_SYSDIVCK_PCLKB_MASK));
    uint32_t pclkb         = (IIC_MASTER_SYSDIVCK_PCLKB_MASK & sysdiccr) >> 8U;
    uint32_t iclk          = sysdiccr >> 24U;
    uint32_t timeout_count = ((1U << (pclkb - iclk)) * p_ctrl->p_reg->ICBRL) >> 2U;

    /* Check if this is a new transaction or a continuation */
    if (!p_ctrl->restarted)
    {
        /* As per BBSY clearing conditions in section 36.2.2 of the RA6M3 manual R01UH0886EJ0100,
         * the BBSY bit is 0 after the bus free time (ICBRL setting)
         * if a start condition is not detected after a stop condition detection.
         */
        IIC_MASTER_HARDWARE_REGISTER_WAIT(p_ctrl->p_reg->ICCR2_b.BBSY, 0U, timeout_count);

        /* If bus is busy, return error */
        // TODO: added to be able to recover from this situation via R_IIC_Abort()
        if (timeout_count == 0) {
            p_ctrl->err = true;
        }
        FSP_ERROR_RETURN((0U != timeout_count), FSP_ERR_IN_USE);

        /* This is not a restarted transaction. Enable TXI for the next transfer.
         * This had been disabled at the end of TXI interrupt.
         * The intention is to only enable IIC_MASTER_TXI_EN_BIT.
         * Writing the whole mask - IIC_MASTER_INTERRUPT_ENABLE_INIT_MASK saves cycles.
         */
        p_ctrl->p_reg->ICIER = IIC_MASTER_INTERRUPT_ENABLE_INIT_MASK;
    }

    /* Initialize fields used during transfer */
    p_ctrl->addr_loaded          = 0U;
    p_ctrl->loaded               = 0U;
    p_ctrl->remain               = p_ctrl->total;
    p_ctrl->addr_remain          = p_ctrl->addr_total;
    p_ctrl->err                  = false;
    p_ctrl->dummy_read_completed = false;
    p_ctrl->activation_on_rxi    = false;
    p_ctrl->activation_on_txi    = false;
    p_ctrl->address_restarted    = false;
[... 42 lines cut ...]
    }
    else
    {
        p_ctrl->restarted = false;
    }

    /*
     * The Flowchart under 36.3.3 "Master Transmit Operation" and section 36.3.4 "Master Receive Operation"
     * of the RA6M3 manual R01UH0886EJ0100 is covered in the interrupts:
     *
     * 1. NACKF processing is handled in the ERI interrupt.
     *    For receive, dummy reading ICDRR is not required because the NACK processing in this driver resets the IIC peripheral.
     * 2. Data is written to ICDRT in the transmit interrupt (TDRE is set when a transmit interrupt fires).
     * 3. For transmit, stop is issued in the transmit end interrupt (TEND is set when a transmit end interrupt fires).
     * 4. For transmit, ICSR2 is cleared in the transmit end interrupt.
     * 5. For receive, remaining processing including reading ICDRR happens in the receive interrupt (RDRF is set when a receive interrupt fires).
     */
    return FSP_SUCCESS;
}

```

Please check if this would be the intended behaviour.

I encountered this situation when I was testing hotplugging i2c slave device, which causes weird behaviour on the bus. And I couldn't recover from it.

SeanMeng0509, 2022-12-14:
Hi @ijager, thank you for the update. We will take a look at this issue. Btw, the original problem was due to "p_ctrl->restart" not reset when IIC communication aborted.

GodavarthyAravinda, 2023-02-24:
Associated https://reasoftware.atlassian.net/browse/FSPRA-363 is done

closed 2023-02-24 by GodavarthyAravinda