renesas/fsp #248: Memory leak in Cellular_Cleanup

RELOC-DEV, 2022-10-25:
The `Cellular_Init` API creates a task that is never deleted by the `Cellular_Cleanup`.

The task creation is in the function `Platform_CreateDetachedThread`
```
file: \ra\fsp\src\rm_cellular_platform_aws\cellular_platform.c
line: 117
```

FSP 4.1.0

renesas-adam-benson, 2022-12-12:
Thanks for the report @RELOC-DEV. Looks like this bug exists inside the Amazon FreeRTOS Cellular Interface. I've created an issue on the GitHub page for that: https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/issues/119

Once there is a fix available for this it should be included in the next FreeRTOS version update we do for FSP.

closed 2022-12-12 by renesas-adam-benson

renesas-adam-benson, 2022-12-14:
@RELOC-DEV the thread exits via a `PKTIO_EVT_MASK_ABORT` event sent when calling `Cellular_Cleanup`. The memory for the thread info is freed in `prvThreadRoutineWrapper` when the thread exits and deletes itself. See the reply in the issue linked above for more info.