renesas/fsp #250: rm_vee_flash.c - segment refresh in RM_VEE_FLASH_Open will fail at start-up

DimionDR, 2022-10-31:
The `rm_vee_refresh_next_data_source` calls `rm_vee_write_seg_hdr` function with empty `seg_hdr.valid_code` field which leads to wrongly formed header on new segment during segment refresh in   `RM_VEE_FLASH_Open`

renesas-billgesner, 2022-11-08:
Hi @DimionDR, this does appear to be a valid issue. I'm investigating the fix and any impacts that it may have.

300bps, 2023-04-29:
Thank you @DimionDR! You appear to have identified and provided a FIX for the issue that was causing periodic virtual EEPROM data loss in my application.

@renesas-billgesner - The pull-request for this fix needs to be accepted ASAP! This bug is triggered by an edge case that is more likely to happen the more segment refreshes a product does. In my application, it results in complete virtual EEPROM data loss. It will have the same impact in other applications.

Discussion:
I have run somewhat extensive testing on the fix provided by @DimionDR, and it does appear to verifiably correct the issue that resulted in an improper segment header being written (the flash segment '.valid_code' location contained 0x0000 instead of the expected 0xBEAD). This improper segment header would cause the virtual EEPROM driver to erase the segment upon a uC reset (or any other event that would cause the 'RM_VEE_FLASH_Open' command to be called).

My application is using 2 segments and no ref data. The virtual EEPROM driver is assigned the entire dataflash.

I was able to trigger the problem in the code (without the fix) in my application with the following steps:

1. Open the virtual EEPROM driver.
2. Fill the segment.
3. Reset the uC. This essentially reboots and reopens the virtual EEPROM driver.
4. Write additional data that triggers a segment refresh
5. Check header in newly written segment.

While not occurring EVERY time, I've been able to semi-reliably duplicate the problem this way.

Virtual EEPROM snippets before and after occurrence of the bug:
<At Address 0x40100800>
<Segment header (first 8 bytes) is CORRECT>
0000 0000 0000 BEAD

RESULT AFTER STEP 5 ABOVE
<At Address 0x40100000>
<Segment header (first 8 bytes) is CORRUPTED>
0001 0000 0000 0000
Note the presence of 0000 in the last position where BEAD should be.

The fix provided by @DimionDR results in BEAD correctly being written in all of the test cases in which I have been able to duplicate the triggering condition.

closed 2023-06-29 by GodavarthyAravinda