renesas/fsp #256: C++ / CMake broken for Smart Configurator

saltyJeff, 2023-02-23:
When I create a new project using the RA Smart Configurator with C++ support enabled and the CMake option selected, the resulting CMakeLists.txt looks like this:

```
# You can call cmake from command line by toolchain file argument   use `-DCMAKE_TOOLCHAIN_FILE=cmake/gcc.cmake`

# Project minimum version
cmake_minimum_required(VERSION 3.16.4)

include(Config.cmake)

enable_language(C ASM)

# Project name
project(asdf VERSION 1.0.0)

include(${CMAKE_CURRENT_LIST_DIR}/cmake/GeneratedSrc.cmake)
```

C++ is not included in the enable_language, even though `src/hal_entry.cpp` is created

When I try to build, the compiler throws the following error:

`'-std=c99' is valid for C/ObjC but not for C++`

Because the output `RASC_CMAKE_CXX_FLAGS` is invalid.

Normally these settings wouldn't be an issue, but all the compiler flags are stored together into 1 giant string in `cmake/GeneratedCfg.cmake`, and it is a hassle having to filter out the useful flags (`-mcpu -D_RA_CORE_`) from the flags that I don't need (`-Wxxx`) and the flags that create a bunch of temporary makefiles.

saltyJeff, 2023-02-28:
RASC version:
```
Smart Configurator

Version: 2022-10 (22.10.0)
Build Id: R20221013-1357
```

FSP version is 4.2.0

renesas-andyb, 2023-03-02:
@saltyJeff  - please can you try with RASC 2023-01, as shipped in FSP 4.3.0. I think this should resolve your issue.

saltyJeff, 2023-03-02:
Hello @renesas-andyb we are currently looking at other mcu offerings so I may be slow to reply in the future.

Found another bug on v4.3.0, you're missing a `SET(CMAKE_CXX_COMPILE_OBJECT "${CMAKE_CXX_COMPILER} ${CMAKE_CXX_FLAGS} -c -o <OBJECT> -x c <SOURCE>")` in `gcc.cmake`

I also noticed there's a new `.vscode/` folder, can you please remove this as it conflicts with many of my pre-sets. If you do want to integrate with vscode-cmake, use the [dedicated `.vscode/cmake-kits.json`](https://github.com/microsoft/vscode-cmake-tools/blob/main/docs/kits.md) instead of the settings file.

One final question, is there a way to make RASC generate code in release-mode? The CXX_FLAGS still has `-g -O0` in the beginning which is set whether the build type is debug or release

renesas-brandon-hussey, 2024-02-06:
Fixes for these issues are currently scheduled for the e2 studio 2024-04 release which should align with the FSP v5.3.0 release.

closed 2024-05-10 by GodavarthyAravinda