renesas/fsp #273: How to submit and obtain third-party packages?

myriadsRays, 2023-06-02:
Hello,
I have two questions about third-party packs : 

1. Has anyone else submitted packages and where can I download these third-party packages?

2. I have developed a pack based on fsp4.3.0 about macronix mx25r flash special command, such as DP command, and through what channels can I submit it for your review? Can I submit my code through pull request in this repository?

thanks.

renesas-brandon-hussey, 2023-06-02:
Hi @myriadsRays and thanks for the question.

> Has anyone else submitted packages and where can I download these third-party packages?

There have been some vendors that have created packs previously. These packs were created to add support for the vendor's product into FSP. These packs were hosted on the vendor's website.

> I have developed a pack based on fsp4.3.0 about macronix mx25r flash special command, such as DP command, and through what channels can I submit it for your review? Can I submit my code through pull request in this repository?

Is this a new FSP module or modifications to existing FSP code?

If it's a new module then my recommendation would be:

1. Put the code in your own GitHub repository
2. Provide that link here. Someone on the FSP team can do a review.
3. Once the review is done, we can provide a link to your repository from this repository. Perhaps an External Contributors section in the README.md.

If this is an update to an existing module, then please create a pull request.

myriadsRays, 2023-07-05:
Hi,
I create a new FSP module for Macronix MX25R nor flash in https://github.com/macronix/fsp/tree/macronix repository.
The source code is in 

> ra/fsp/inc/api/r_mx25r_extend_api.h
> ra/fsp/inc/instances/r_mx25r_extend.h
> ra/fsp/src/r_mx25r/r_mx25r_extend.c

The module description and toolingSupport.xml are in 

> scripts/xml/mx25r_extend/

The document is in

> doc/mx25r_extend

The vendor's link is https://www.macronix.com/en-us/support/design-support/Pages/software-support.aspx
I hope your team can give me some reviews.

myriadsRays, 2023-07-11:
Hi,

We are continuously developing FSP packs tailored for Macronix products. We have recently developed a new FSP module pack for the Macronix MX25R NOR Flash, which can be found in the https://github.com/macronix/fsp/tree/macronix/ repository.

Based on customer feedback, they have encountered some issues when using the MX25R NOR Flash on the FSP platform. The MX25R NOR Flash includes specific commands like DEEP_POWER_DOWN, RELEASE_DEEP_POWER_DOWN, HIGH_PERFORMANCE_MODE, and LOW_POWER_MODE. However, the FSP does not currently provide corresponding command operations.

To address this, we have developed a module called MXIC.MX25R_Extend.1.0.0+fsp.4.3.0.pack that implements the functionalities of MX25R_HighPerformanceMode, MX25R_LowPowerMode, MX25R_Enter_DeepPower, and MX25R_Exit_DeepPower. Our aim is to accelerate our customers' development progress.

This module primarily provides the following APIs:
```
fsp_err_t setNormalMode();
fsp_err_t setHighPerformanceMode();
fsp_err_t setLowPowerMode();
fsp_err_t setSuperLowPowerMode();
fsp_err_t setDeepPowerDownMode();
fsp_err_t power_init();
```
The source code of this pack can be found in the following locations:

> ra/fsp/inc/api/r_mx25r_extend_api.h
> ra/fsp/inc/instances/r_mx25r_extend.h
> ra/fsp/src/r_mx25r/r_mx25r_extend.c

The module description and tooling support can be found in:

> script/xml/mx25r_extend

The documentation is located at:

> doc/mx25r_extend

For more information, you can visit the vendor's website at https://www.macronix.com/en-us/support/design-support/Pages/software-support.aspx.

We kindly request your team to review this work.

Thanks.