renesas/fsp #275: Linker Error Related to Veneer$$CMSE

pltmcu-jci, 2023-06-12:
**Details**:
FSP Version: v4.4.0
Target: Renesas RA6M5
Toolchain: IAR v9.20.4.47112

**Issue Description**
After upgrading the FSP from version 3.8.0 to version 4.4.0, I started getting a warning from my IAR project for the RA6M5 target.

`Warning[Lp051]: some sections in automatic block "Veneer$$CMSE" would have been placed differently if they were not part of an automatic block. Example: section "Veneer$$CMSE (main_secure.o #71)" would have matched "section Veneer$$CMSE" at line 83 of "C:\Firmware\Auto_Repos\plt-mcu-product\product\large\carrier-test\Make\main_secure\..\..\..\..\..\ide\linker\large\main_secure_renesas.icf". Instead, it is placed in block "Veneer$$CMSE" which matched "section Veneer$$CMSE" at line 83 of "C:\Firmware\Auto_Repos\plt-mcu-product\product\large\carrier-test\Make\main_secure\..\..\..\..\..\ide\linker\large\main_secure_renesas.icf"`

The line 83 being referenced in the warning is `place in NSC_region    { section .tz_flash_nsc_start, section Veneer$$CMSE };`

The `Veneer$$CMSE` is defined here https://github.com/renesas/fsp/blob/master/ra/fsp/src/bsp/mcu/all/bsp_security.c#L64

Note that the warning goes away if I update the line in the Linker Script to: `place in NSC_region    { section .tz_flash_nsc_start, section "section Veneer$$CMSE" };`

renesas-jbrabender, 2023-06-12:
@pltmcu-jci ,
We apologize for this warning. It is being addressed.
In meantime it can be corrected by changing this line in the fsp.icf file

_define block Image$$__tz_FLASH_C with alignment = __tz_FLASH_C_ALIGN { section Veneer$$CMSE* };_

Please note the "*" immediately after the CMSE
For reference, although this warning was generated, the linker output is the same with our without the "*" (no binary change by this modification)

renesas-jbrabender, 2023-06-12:
@pltmcu-jci ,
Sorry, but you're obviously not using our FSP linker scripts, (above is for our provided scripts)...but I'm sure you understand the modification point.

pltmcu-jci, 2023-06-13:
@renesas-jbrabender Thanks for the prompt response. This is confirmation that the warning has been resolved.

closed 2023-06-13 by pltmcu-jci