renesas/fsp #284: Configuring 8 Mbit/s CAN-FD bitrate using TSEG1=2 and TSEG2 = 2 causes assertion to fail

hwmaier, 2023-08-11:
The assertion check in r_canfd_bit_timing_parameter_check() (which would be correct for nominal bit rates) is wrongly applied to CAN-FD data rates as well and causes the assertion to fail if CAN-FD data rates are configured which require TSEG1 and TSEG2 to be set to the same value. 

The datasheet shows different conditions for nominal (`>`) and data bitrates (`>=`) which the assertion check should take into account:

[image]

This is relevant for CAN-FD because for example the CAN-FD bit rate of 8 Mbit/s requires TSEG1 = 2 and TSEG2 = 2 to be set. This is the only setting available to achieve  8 MBit/s and there is no workaround.

[image]

Refer to this line of code which needs to be changed:
https://github.com/renesas/fsp/blame/d52e5a6a59b7c638da860c2bb309b6e78e752ff8/ra/fsp/src/r_canfd/r_canfd.c#L879

Refer to section 32.4.1.1 "Bit Timing Conditions" in the RA6M5 User's Manual.

jprofeta-rea, 2023-08-16:
Thanks for reporting this issue, @hwmaier. I've verified it and have started working on the fix.

renesas-austin-hansen, 2023-09-01:
The fix for this issue will be available in the upcoming FSP 5.0.0 release.  Thanks again for reporting this!

closed 2023-10-27 by GodavarthyAravinda