renesas/fsp #298: No NULL-Pointer check in gpt_counter_underflow_isr in r_gpt.c

stefannagelchargebyte, 2023-10-24:
Dear Renesas Team,
I'm using the gpt_counter_underflow_isr  which is linked to an ADC conversion.
However, even if I use the gpt_counter_underflow_isr, I dont use the callback. This leads to a critical assert.

I would propose as you did for the gpt_counter_overflow_isr only add a check against NULL pointer for the callback.

Actual code:
`void gpt_counter_underflow_isr (void)
{
    /* Save context if RTOS is used */
    FSP_CONTEXT_SAVE

    IRQn_Type irq = R_FSP_CurrentIrqGet();

    /* Clear pending IRQ to make sure it doesn't fire again after exiting */
    R_BSP_IrqStatusClear(irq);

    /* Recover ISR context saved in open. */
    gpt_instance_ctrl_t * p_instance_ctrl = (gpt_instance_ctrl_t *) R_FSP_IsrContextGet(irq);

    /* Call user callback. */
    r_gpt_call_callback(p_instance_ctrl, TIMER_EVENT_TROUGH, 0);

    /* Restore context if RTOS is used */
    FSP_CONTEXT_RESTORE
}`

Proposal:
`void gpt_counter_underflow_isr (void)
{
    /* Save context if RTOS is used */
    FSP_CONTEXT_SAVE

    IRQn_Type irq = R_FSP_CurrentIrqGet();

    /* Clear pending IRQ to make sure it doesn't fire again after exiting */
    R_BSP_IrqStatusClear(irq);

    /* Recover ISR context saved in open. */
    gpt_instance_ctrl_t * p_instance_ctrl = (gpt_instance_ctrl_t *) R_FSP_IsrContextGet(irq);

    /* Call user callback. */
    if (NULL != p_instance_ctrl->p_callback)
    {
        r_gpt_call_callback(p_instance_ctrl, TIMER_EVENT_TROUGH, 0);
    }

    /* Restore context if RTOS is used */
    FSP_CONTEXT_RESTORE
}`

renesas-brandon-hussey, 2023-10-24:
Hi @stefannagelchargebyte why are you enabling the GPT underflow interrupt if you are not going to have a callback? If you are using the GPT underflow event to trigger an ADC conversion using the ELC then you do not need to enable the GPT underflow interrupt.

stefannagelchargebyte, 2023-10-25:
Because I didn't understand it :-)
I thought you need the interrupt to trigger the event ... but after your mail I disabled the interrupts and the events are coming through, anyway.
Thanks for the hint

renesas-brandon-hussey, 2023-10-25:
No problem, glad to hear it solved your problem!

closed 2023-10-25 by renesas-brandon-hussey