renesas/fsp #329: Hardware TRNG draws excessive current after use when placed in power down.

dsmith9000, 2023-11-01:
### Problem Description
The FSP code that powers down the SCE / TRNG module in the 'HW_SCE_PowerOff' function does not disable the hardware random number generator (TRNG) before stopping the module clock to it. This can leave the TRNG enabled and drawing an additional 370uA (as measured with a Joulescope in our application) when powered down. Note that the TRNG is only enabled and made active by the FSP if BOTH of the following occur:

1. Call 'HW_SCE_PowerOn' (or the higher level 'HW_SCE_McuSpecificInit') to enable the module clock to the TRNG.
2. Make at least 1 call to 'HW_SCE_RNG_Read' to read a random sequence.

After these steps, calling 'HW_SCE_PowerOff' to power down the SCE and TRNG leaves the TRNG enabled when the module clock is stopped. This causes the microcontroller to consume an additional 370uA continuously.

### Fix
The fix (implemented in pull request #328) is to disable the TRNG in 'HW_SCE_PowerOff' BEFORE the module clock is stopped.

renesas-brandon-hussey, 2024-01-19:
This is being internally tracked using FSPRA-2352.

kenvinh, 2024-07-29:
Is there any update on this issue? We're having the exact same problem, after tracking for few hours, and apply similar patch to @dsmith9000  then can see the current drop to expected value

Please help us to follow up on this so we can reduce number of custom code

michaelthomasj, 2024-07-29:
@kenvinh , this will be in the FSP 5.5 release scheduled for August.

kenvinh, 2024-07-30:
@michaelthomasj, appreciate for clear milestone, thank you.

PR #328 (2024-07-30): Bugfix: Disable the hardware TRNG before stopping the module clock to it to prevent excessive current draw in power down.
### Problem Description
The FSP code that powers down the SCE / TRNG module in the 'HW_SCE_PowerOff' function does not disable the hardware random number generator (TRNG) before stopping the module clock to it.  This can leave the TRNG enabled and drawing an additional 370uA (as measured with a Joulescope in our application) when powered down.  Note that the TRNG is only enabled and made active by the FSP if BOTH of the following occur: 
1) Call 'HW_SCE_PowerOn' (or the hig [... cut]