renesas/fsp #344: BSP_CFG_HOCO_FREQUENCY is only used if there is FLL

metebalci, 2024-01-21:
I am trying the new Arduino R4 board which has RA4M1. While dumping clock related registers after restart, I realized `HOCOCR2` is `0 (24 MHz)` although `BSP_CFG_HOCO_FREQUENCY` is defined as `4 (48 Mhz)`. Looking at startup codes, as far as I can see `BSP_CFG_HOCO_FREQUENCY` is only used if `BSP_PRV_HOCO_USE_FLL` is set, and this is set only if `BSP_FEATURE_CGC_HAS_FLL`, `BSP_CFG_FLL_ENABLE` and `BSP_CLOCK_CFG_SUBCLOCK_POPULATED` is set. However, `BSP_FEATURE_CGC_HAS_FLL` is set to 0 in `ra4m1/bsp_feature.h`, thus, I think, `BSP_CFG_HOCO_FREQUENCY` has no effect so HOCO frequency is not correctly updated during startup. I think the CLKOUT pin is not exposed in this board, so I cannot verify the actual HOCO frequency. Am I missing something or `HOCOCR2` should be set to `BSP_CFG_HOCO_FREQUENCY` during startup ?

(side not: HOCOCR2 is not defined in R_SYSTEM, #343)

renesas-brandon-hussey, 2024-01-22:
This is being internally tracked using FSPRA-2363.

renesas-colin-hines, 2024-04-16:
@metebalci:
1. Can you please provide the version of FSP and/or Arduino library that you are using?
2. Can you confirm the address that you are checking for HOCOCR2. Please note that the address given in the current Hardware Manual (r01uh0887ej0110) - 0x4001E000 - is **NOT** correct. The correct address is **0x4001E037**, as indicated in the Technical Note adding it originally: https://www.renesas.com/us/en/document/tcu/addition-hococr2-register-hoco-clock-control
3. Please note that the FLL relationship you were seeing in code is not how FSP configures the HOCO frequency; it is for configuring the FLL functionality. The HOCO frequency is configured via the OFS1 register in ROM and HOCOCR2 will be initialized to match that setting at power on reset. See https://github.com/renesas/fsp/blob/master/ra/fsp/src/bsp/mcu/all/bsp_rom_registers.c#L32
4. Lastly, please check the hex file contents for your program and provide the value located at 0x0404 - 0x0407. This should be the contents of the OFS1 value.

renesas-tam-dinh, 2025-02-08:
Hi @metebalci ,
Could you please let me know if there is any update at your side?

renesas-brandon-hussey, 2025-03-17:
Closing this ticket as there has been no reply. Please reopen as needed.

closed 2025-03-17 by renesas-brandon-hussey