renesas/fsp #346: RA8 default clock configuration provides incorrect PLL input frequency

renesas-austin-hansen, 2024-02-29:
## Issue
The default clock configuration provides PLL input clock frequencies (after division) that are out-of-range.  The allowable range is 6-12 MHz.

## Workaround
Adjust the PLL input divisor such that the resulting frequency is between 6-12 MHz.

Examples:
- PLL source is HOCO at 48 MHz -> Set input divisor to /4 (12 MHz)
- PLL source is MOSC at 20 MHz -> Set input divisor to /2 (10 MHz)

xianghui-renesas, 2024-02-29:
Hi @renesas-austin-hansen , 
It seems when migrating to FSP v5.2.0 and when working with the EKs, the following updates are needed. Could you confirm?
- change PLL Div/1 to PLL Div/2
- change PLL Mulx48.00 to PLL Mulx96.00
If there is customized clock for additional peripherals other than the EK pin config defaults, will the system keep the updates when migrating to FSP v5.2.0 by design?  
Thanks for any comments you may have.

renesas-austin-hansen, 2024-02-29:
Hi @xianghui-renesas, this is not a matter of FSP versions but rather hardware specification.  All projects should ensure the PLL input frequency post-division is 6-12 MHz.  The clock settings are not updated when changing FSP versions so the user will either need to click Restore Defaults or manually update the settings.

That said, for EK projects using the default settings, changing to /2 x96 is enough and matches the new default.