renesas/fsp #347: Crash when calling RM_VEE_FLASH_Open

sytsereitsma, 2024-03-05:
Hi,

We have the following scenario:
* A bootloader with vee flash configured for a max record id of 32 (only reads data from flash)
* An application with vee flash configured for a max record id of 64

When opening the flash from the bootloader (`RM_VEE_FLASH_Open`) we sometimes get a crash, which I narrowed down to the record offset assignment in `rm_vee_load_record_table` (line 1055 on v5.2.0):
```c
        /* Save record offset if complete record found (reset may have occurred during a write) */
        if (RM_VEE_FLASH_VALID_CODE == p_end->valid_code)
        {
            // This line may assign an offset out of bounds of the rec_offset array
            p_ctrl->p_cfg->rec_offset[p_end->id] = (uint16_t) (addr - p_ctrl->active_seg_addr);

            /* Save for statusGet */
            p_ctrl->last_id = p_end->id;
        }
```
When parsing all records sometimes record id's (`p_end->id`) that are greater than 32 resulting in an out of bounds write to the `p_ctrl->p_cfg->rec_offset` array.

A quick fix would be:
```c
        /* Save record offset if complete record found (reset may have occurred during a write) */
        if (RM_VEE_FLASH_VALID_CODE == p_end->valid_code)
        {
            uint16_t rec_id = p_end->id;
            if (rec_id <= p_ctrl->p_cfg->record_max_id)
            {
                p_ctrl->p_cfg->rec_offset[p_end->id] = (uint16_t) (addr - p_ctrl->active_seg_addr);

                /* Save for statusGet */
                // Not sure if this should be inside, or outside the if
                p_ctrl->last_id = p_end->id;
            }
        }
```

renesas-brandon-hussey, 2024-03-05:
This is being internally tracked using FSPRA-2635.

renesas-billgesner, 2024-03-13:
Hi @sytsereitsma, opening a volume with RM_VEE_Open with a configuration that does not match what is reflected in the volume is not currently supported. Is there a reason you cannot open the volume using the same configuration in both the bootloader and the application?

sytsereitsma, 2024-03-18:
At present we have no means to field-update our bootloader.
In the beginning we thought 32 records 'ought to be enough for anybody', so now we're stuck with that number. Meanwhile the application has evolved and now uses a max record id of 64 entries.