renesas/fsp #353: Using SCE5 in a C++project causes a build error

dozen, 2024-05-05:
The issue reported in #132 is also occurring in SCE5. The header file required to use the SCE5 module (r_sce_if.h) contains C++ reserved keywords, which prevents it from being built with a C++ compiler.

I believe the solution or migration will be similar to what is described in #132.

### Reference
- `ra/fsp/src/r_sce/crypto_procedures/src/sce5/plainkey/public/inc/r_sce_if.h` https://github.com/renesas/fsp/blob/d5e6aff684490605063cf5289bad0531cffdc175/ra/fsp/src/r_sce/crypto_procedures/src/sce5/plainkey/public/inc/r_sce_if.h#L594-L598
- #132

renesas-brandon-hussey, 2024-05-06:
This is being internally tracked using FSPRA-3016.

michaelthomasj, 2024-05-10:
@dozen , 
Thank you for pointing out this issue. The fix for this issue will be in our next release scheduled for the end of this month. 
-Michael

closed 2024-06-27 by GodavarthyAravinda