renesas/fsp #363: Is line 112 of bsp_irq.c supposed to be "NVIC->IPR" instead of "NVIC->IP"?

jodyono, 2024-07-03:
Just installed FSP 5.4.0 and am getting a build error on my project where the compiler says that there is no "IP" field for "NVIC" at line 112 of bsp_irq.c.  Was it supposed to be "NVIC->IPR[...]"?

Here's the error:

..\ra\fsp\src\bsp\mcu\all\bsp_irq.c:122:11: error: 'NVIC_Type' has no member named 'IP'; did you mean 'IPR'?
  122 |     NVIC->IP[((uint32_t) irq)] = (uint8_t) ((priority << (8U - __NVIC_PRIO_BITS)) & (uint32_t) UINT8_MAX); // JPO: Was "NVIC->IP"
      |           ^~
      |           IPR

renesas-austin-hansen, 2024-07-03:
This is due to a compatibility break in the Cortex-M registers provided in CMSIS 6.  These registers were previously called `IP` for CM4 devices, but have been renamed to `IPR` to match later cores.  This issue was found just after finalizing the v5.4.0 release, but it has been fixed in our internal development repo.

It looks like you have probably already edited `bsp_irq.c` to resolve the issue.  If you prefer not to edit FSP code, enabling the `Enable inline BSP IRQ functions` setting in the BSP properties will use different ICU code that does not exhibit this issue.

jodyono, 2024-07-03:
Thanks much for the full explanation!

Best regards,
J.O.

closed 2024-07-03 by jodyono