renesas/fsp #364: R_GPT_InfoGet only reports count direction TIMER_DIRECTION_UP

Michael812498, 2024-07-04:
The R_GPT_InfoGet function appears to be hard coded to fill timer_info_t with TIMER_DIRECTION_UP
https://github.com/renesas/fsp/blob/bd5dbafc705cd53c83a6c37114fb3f08fd14f25a/ra/fsp/src/r_gpt/r_gpt.c#L683-L684

My interpretation of the intent of the function should be that it should respond with:

`p_info->count_direction = p_instance_ctrl->p_reg->GTUDDTC_b_UD;` or something to that effect as it is possible to configure the GPT to count down.

renesas-austin-hansen, 2024-07-08:
Hi Michael, currently the RA FSP driver for GPT only supports up-counting and so this value is hardcoded.  It is possible to configure the hardware to down-count mode but we do not currently expose this functionality.

One further note - GPT does automatically switch direction back and forth in triangle PWM modes.  I do not think providing the actual count direction would be useful in this case given polling would be very inexact for motor control (versus events or interrupts), but a case could be made to provide it regardless if there are strong opinions.

Michael812498, 2024-07-08:
In my case I have a cascade timer/counter setup to gauge pulse outputs to absolute counts. As you say I currently manage UD through interruption by register access for timeliness, but encountered this as I was using the Info API as a part of some of the reporting and monitoring functions that have no specific realtime demands.

I have no strong preferences, the API is theoretically more portable but as you point out the argument is kind of contrived considering that you are likely operating outside of the API to some capacity if you are interested in the GPT's UD at this time.

renesas-austin-hansen, 2024-10-08:
I've gone back and forth on this some.  Ultimately because this is related to `R_GPT_InfoGet`, which is intended to relay configuration and not status, I do not think it makes sense to return the current actual count direction in triangle modes.  If we supported this info via `R_GPT_StatusGet` I think there is a stronger case, but direction is not one of the elements currently provided in `timer_status_t` and a use case is unlikely if it were to be added.

closed 2024-10-08 by renesas-austin-hansen