renesas/fsp #369: MIPI DSI (R_MIPI_DSI_Command) - Asynchronous commands don't transmit for all virtual channels

JGrenard, 2024-08-23:
## Issue
Transmission of GLCDC data is not affected

Transmission of LP asynchronous messages - sent via R_MIPI_DSI_Command - will only output when the command channel is set to virtual channel (VC) 0.

Transmission of HS asynchronous messages- sent via R_MIPI_DSI_Command - will only output when the command channel is set to VC 1.
Note: This means that LP commands will be received only by displays configured as VC 0 (the most common configuration) and HS commands will be received only by displays configured as VC 1.

## Workaround
This issue does not affect situations where exclusively LP asynchronous commands are sent to a display controller configured for use of virtual channel 0 (this is common). Similarly, this issue would not affect configurations where exclusively HS asynchronous commands are sent to a display controller that uses virtual channel 1 (not common). For other configurations, there is no workaround at this time.

renesas-abigail, 2024-10-09:
This issue is tracked internally by FSPRA-3464.

renesas-abigail, 2024-10-09:
This issue was fixed in FSP v5.5.0.