renesas/fsp #375: Unable to select CANFDCLK source as XTAL in RA FSP v5.5.0 on MacOS

nniranjhana, 2024-10-15:
Hi there, I have imported existing configuration.xml from a team repo in my e2 studio with FSP for RA family 5.5.0 on MacOS. My team members use Linux and FSP 4.5.0, and is the project's current version as well.

When importing the project, I get errors on startup and I cannot follow the same configuration (CANFDCLK source as XTAL), I have to switch it to the available sources (PLL and PLL2) to be able to continue and generate further configuration. I believe it is a bug in the current version that I use. Could you confirm this?

**More details:**
This is in our project config (works for others on FSP 4.5.0):
`<node id="board.clock.canfdclk.source" option="board.clock.octaspiclk.source.xtal"/>`
which I can't use because of unavailable option here:
[image]

omarQureshiRenesas, 2024-10-16:
Which MCU was the original project built for? Is the MCU target of the new FSP 5.5.0 project the same?

omarQureshiRenesas, 2024-10-16:
According the hardware user's manual, the CANFDCLK can only be PLL or PLL2 (see CANFDCKSEL[2:0]) .  Therefore there was an update made within FSP to remove inaccurate selections from the CANFDCLK selections.  FSP v5.5.0 correctly matches hardware functionality.  Reference: RA6M5 Hardware Users Manual Rev 1.30. https://www.renesas.com/en/document/man/ra6m5-group-users-manual-hardware?r=1493931 .     Please review and verify, then close this ticket.

nniranjhana, 2024-10-16:
Correct, thanks, closing.

closed 2024-10-16 by nniranjhana