renesas/fsp #386: Invalid PLL2 configuration in R_CGC_ClocksCfg FSP5.7.0

andross092, 2025-01-08:
Please check line 511 in r_cgc.c

the code   

 `       err =
            r_cgc_pllccr_pll_hz_calculate(&p_clock_cfg->pll2_cfg, CGC_CLOCK_PLL2, pll2_hz, &pllccr);`
uses the variable `pllccr` filled for PLL configuration.
It should be `pll2ccr`
Thanks for support

renesas-austin-hansen, 2025-01-08:
Hi @andross092, which version of FSP are you using?  I was not able to find this issue in recent versions.

andross092, 2025-01-08:
I'm using FSP 5.7.0, 
anyway it is in the master branch
 [https://github.com/renesas/fsp/blob/master/ra/fsp/src/r_cgc/r_cgc.c#L598](url)

renesas-austin-hansen, 2025-01-08:
Ah sorry, line numbers are a bit offset in our development system and I got mixed up with a similar function.  We will investigate.