renesas/fsp #387: BSP_FEATURE_FLASH_HP_SUPPORTS_DUAL_BANK causes g_bsp_rom_banksel_sel to be missing on RA4M3

pyhalte, 2025-01-10:
`BSP_FEATURE_FLASH_HP_SUPPORTS_DUAL_BANK` condition prevents **BANKSEL_SEL** register from being included in hex file with RA4M3.

FSP 4.1.0 worked correctly, but at least 4.6.0 and 5.7.0 do have this bug.

I believe following if condtion should not exist around this code in `bsp_rom_registers.c`:

```
   #if BSP_FEATURE_FLASH_HP_SUPPORTS_DUAL_BANK
BSP_DONT_REMOVE static const uint32_t BSP_PLACE_IN_SECTION(".option_setting_banksel_sel") g_bsp_rom_banksel_sel =
    BSP_CFG_ROM_REG_BANKSEL_SEL;

   #endif
```

For comparison, **BANKSEL_SEC** register doesn't have this condition, and is included in hex file correctly:

```
BSP_DONT_REMOVE static const uint32_t BSP_PLACE_IN_SECTION(".option_setting_banksel_sec") g_bsp_rom_banksel_sec =
    0xFFFFFFFF;
```

Condition around `g_bsp_rom_dualsel` definition also removes register, but I believe this is correct behaviour as RA4M3 manual doesn't mention register existing at 0x0100_A110.

```
   #if BSP_FEATURE_FLASH_HP_SUPPORTS_DUAL_BANK
BSP_DONT_REMOVE static const uint32_t BSP_PLACE_IN_SECTION(".option_setting_dualsel") g_bsp_rom_dualsel =
    BSP_CFG_ROM_REG_DUALSEL;

   #endif
```

I think there has been a mistake between dual bank feature, which RA4M3 doesn't have, and bank selection feature, which RA4M3 has.

pyhalte, 2025-01-10:
Here's what Option-Setting Memory registers look like in hex file with FSP 4.1.0:

```
:020000040100F9
:04A100003DBFFFFF61
:04A13400FFFFFFFF2B
:04A20000FBFDFFFF64
:04A21000FFFFFFFF4E
:0CA24000FFFFFFFFFFFFFFFFFFFFFFFF1E
:0CA26000FFFFFFFFFFFFFFFFFFFFFFFFFE
:04A28000F8F8FFFFEC
:04A29000FFFFFFFFCE
:0CA2C000FFFFFFFFFFFFFFFFFFFFFFFF9E
```

And with FSP 5.7.0:

```
:020000040100F9
:04A100003DBFFFFF61
:04A13400FFFFFFFF2B
:04A20000FBFDFFFF64
:04A21000FFFFFFFF4E
:10A24000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1E
:10A26000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
:04A28000F8F8FFFFEC
:10A2C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9E
```

Notice how second to last row is missing.

renesas-billgesner, 2025-01-13:
This is being internally tracked using FSPRA-4315.