renesas/fsp #388: USB interrupt race condition?

MoreTore, 2025-03-05:
Using the usb_pcdc no-rtos example, the program stops responding to usb input after printing lots of data to the console.  This issue is present on fsp version 3.0.0 and latest 5.8.0. There seems to be a race condition with how the interrupt registers are being handled.

Reproduce with this commit:
https://github.com/MoreTore/ra-fsp-examples/commit/9db1fde0230c9cc6eaa29d35924492cf0d1cb23b

example_projects/ek_ra2a1/usb_pcdc/usb_pcdc_ek_ra2a1_ep/e2studio/src/hal_entry.c

renesas-austin-hansen, 2025-03-05:
Hi @MoreTore, are you getting any errors from USB calls with Parameter Checking enabled or do they always return `FSP_SUCCESS`?

MoreTore, 2025-03-05:
> Hi @MoreTore, are you getting any errors from USB calls with Parameter Checking enabled or do they always return `FSP_SUCCESS`?

No errors.

MoreTore, 2025-03-06:
Work around:

```c
static fsp_err_t check_for_write_complete(void)
{
    usb_status_t usb_write_event = USB_STATUS_NONE;
    int32_t timeout_count = UINT16_MAX;
    fsp_err_t err = FSP_SUCCESS;

    do
    {
        err = R_USB_EventGet (&g_basic0_ctrl, &usb_write_event);
        if (FSP_SUCCESS != err)
        {
            return err;
        }

        --timeout_count;

        if (0 > timeout_count)
        {
            timeout_count = 0;
            err = (fsp_err_t)USB_STATUS_NONE;
            break;
        }

        if (USB_STATUS_READ_COMPLETE == usb_write_event) { //  <-- need to check for USB_STATUS_READ_COMPLETE and read the data so it doesn't get stuck with interrupts off
            R_USB_Read (&g_basic0_ctrl, g_buf, 1, (uint8_t)g_usb_class_type);
        }
    }while(USB_STATUS_WRITE_COMPLETE != usb_write_event);

    return err;
}

MoreTore, 2025-03-06:
Ok so with the above work around its hard to tell if this is a bug in the example or a bug in the driver design or a little bit of both. Polling R_USB_EventGet  in check_for_write_complete and then ignoring events that are not USB_STATUS_WRITE_COMPLETE seems like a bad implementation based on the driver design. R_USB_EventGet is doing a lot more than just getting an event, its doing the USB task and if you choose to ignore what its doing then bad stuff will happen.

rautonkar, 2025-03-06:
@MoreTore This is mostly an issue with how the application is written, which could have been avoided with some documentation of how R_USB_EventGet is supposed to work. Ideally, in the application, you'd check a flag variable and then do a USB_Write if that flag is cleared. Flag clearing would be done when the superloop in hal_entry.c when it notifies a USB_STATUS_WRITE_COMPLETE event.

The improvement in documentation should be that R_USB_EventGet should be invoked from one location only as registered events are removed from the queue upon exit.

MoreTore, 2025-03-06:
Yea that was my conclusion as well. The example should be updated to reflect this.

MoreTore, 2025-03-07:
> [@MoreTore](https://github.com/MoreTore) This is mostly an issue with how the application is written, which could have been avoided with some documentation of how R_USB_EventGet is supposed to work. Ideally, in the application, you'd check a flag variable and then do a USB_Write if that flag is cleared. Flag clearing would be done when the superloop in hal_entry.c when it notifies a USB_STATUS_WRITE_COMPLETE event.
> 
> The improvement in documentation should be that R_USB_EventGet should be invoked from one location only as registered events are removed from the queue upon return from R_USB_EventGet .

From my understanding, USB_STATUS_WRITE_COMPLETE occours when the buffer empty interrupt status register for a port equals 1. Maybe it would be good to just make a blocking usb write call?