renesas/fsp #408: CLion cant open RASC Cmake project correctly

Mr-P-hainan, 2025-07-09:
**I generate a new Cmake project with arm-gcc by using RASC, then i use CLion to open the folder as a cmake project. But it seems that my CLion and Toolchain didnt recognize , or the project is lack for some files? How can i fix this?
My FSP version is 6.0.0 ,RASC version is 2025.04.1**

**Heres is the error_info that cmake gives:**

"D:\PersonalSoftware\CLion 2025.1.1\bin\cmake\win\x64\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_MAKE_PROGRAM=D:/Compiler/mingw64/bin/mingw32-make.exe -DCMAKE_C_COMPILER=D:/Compiler/arm-eabi-gcc13.3.1/bin/arm-none-eabi-gcc.exe -DCMAKE_CXX_COMPILER=D:/Compiler/arm-eabi-gcc13.3.1/bin/arm-none-eabi-g++.exe -G "MinGW Makefiles" -S E:\WorkingDir\FSPSmartConfigurator\FSP_Project -B E:\WorkingDir\FSPSmartConfigurator\FSP_Project\cmake-build-debug
CMAKE_CURRENT_BINARY_DIR is defined as E:/WorkingDir/FSPSmartConfigurator/FSP_Project/cmake-build-debug
Single-config generator(MinGW Makefiles) is active
CMAKE_BUILD_TYPE is defined as Debug
CMAKE_RUNTIME_OUTPUT_DIRECTORY is defined as 
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- The ASM compiler identification is GNU
-- Found assembler: D:/Compiler/arm-eabi-gcc13.3.1/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: D:/Compiler/arm-eabi-gcc13.3.1/bin/arm-none-eabi-gcc.exe
-- Check for working C compiler: D:/Compiler/arm-eabi-gcc13.3.1/bin/arm-none-eabi-gcc.exe - broken
CMake Error at D:/PersonalSoftware/CLion 2025.1.1/bin/cmake/win/x64/share/cmake-3.31/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "D:/Compiler/arm-eabi-gcc13.3.1/bin/arm-none-eabi-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: 'E:/WorkingDir/FSPSmartConfigurator/FSP_Project/cmake-build-debug/CMakeFiles/CMakeScratch/TryCompile-045z9y'
    
    Run Build Command(s): "D:/PersonalSoftware/CLion 2025.1.1/bin/cmake/win/x64/bin/cmake.exe" -E env VERBOSE=1 D:/Compiler/mingw64/bin/mingw32-make.exe -f Makefile cmTC_1929f/fast
    D:/Compiler/mingw64/bin/mingw32-make.exe  -f CMakeFiles\cmTC_1929f.dir\build.make CMakeFiles/cmTC_1929f.dir/build
    mingw32-make[1]: Entering directory 'E:/WorkingDir/FSPSmartConfigurator/FSP_Project/cmake-build-debug/CMakeFiles/CMakeScratch/TryCompile-045z9y'
    Building C object CMakeFiles/cmTC_1929f.dir/testCCompiler.c.obj
    D:\Compiler\arm-eabi-gcc13.3.1\bin\arm-none-eabi-gcc.exe   -fdiagnostics-color=always -o CMakeFiles\cmTC_1929f.dir\testCCompiler.c.obj -c E:\WorkingDir\FSPSmartConfigurator\FSP_Project\cmake-build-debug\CMakeFiles\CMakeScratch\TryCompile-045z9y\testCCompiler.c
    Linking C executable cmTC_1929f
    "D:\PersonalSoftware\CLion 2025.1.1\bin\cmake\win\x64\bin\cmake.exe" -E cmake_link_script CMakeFiles\cmTC_1929f.dir\link.txt --verbose=1
    D:\Compiler\arm-eabi-gcc13.3.1\bin\arm-none-eabi-gcc.exe CMakeFiles/cmTC_1929f.dir/testCCompiler.c.obj -o cmTC_1929f
    D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1\libc.a(libc_a-exit.o): in function `exit':
    exit.c:(.text.exit+0x28): undefined reference to `_exit'
    D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1\libc.a(libc_a-closer.o): in function `_close_r':
    closer.c:(.text._close_r+0x18): undefined reference to `_close'
    D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1\libc.a(libc_a-lseekr.o): in function `_lseek_r':
    lseekr.c:(.text._lseek_r+0x24): undefined reference to `_lseek'
    D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1\libc.a(libc_a-readr.o): in function `_read_r':
    readr.c:(.text._read_r+0x24): undefined reference to `_read'
    D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1\libc.a(libc_a-writer.o): in function `_write_r':
    writer.c:(.text._write_r+0x24): undefined reference to `_write'
    D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: D:/Compiler/arm-eabi-gcc13.3.1/bin/../lib/gcc/arm-none-eabi/13.3.1\libc.a(libc_a-sbrkr.o): in function `_sbrk_r':
    sbrkr.c:(.text._sbrk_r+0x18): undefined reference to `_sbrk'
    collect2.exe: error: ld returned 1 exit status
    mingw32-make[1]: *** [CMakeFiles\cmTC_1929f.dir\build.make:101: cmTC_1929f] Error 1
    mingw32-make[1]: Leaving directory 'E:/WorkingDir/FSPSmartConfigurator/FSP_Project/cmake-build-debug/CMakeFiles/CMakeScratch/TryCompile-045z9y'
    mingw32-make: *** [Makefile:133: cmTC_1929f/fast] Error 2
    
  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:36 (project)

-- Configuring incomplete, errors occurred!

renesas-andyb, 2025-07-22:
@Mr-P-hainan 
I'm glad to hear that you managed to get things working.
CLion isn't currently an environment we directly test/support, though obviously as it consumes cmake files, we would hope you could get projects building without too many issues.
Was the environment variable you needed to set up something other than the setup described in the RASC Quick Start Guide (available on the Help menu within RASC) or the readme that RASC generates into cmake projects ?

[image]

Regards, Andy

Mr-P-hainan, 2025-07-22:
> [@Mr-P-hainan](https://github.com/Mr-P-hainan) I'm glad to hear that you managed to get things working. CLion isn't currently an environment we directly test/support, though obviously as it consumes cmake files, we would hope you could get projects building without too many issues. Was the environment variable you needed to set up something other than the setup described in the RASC Quick Start Guide (available on the Help menu within RASC) or the readme that RASC generates into cmake projects ?
> 
> <img alt="Image" width="562" height="335" src="https://private-user-images.githubusercontent.com/96128724/469019088-59f76005-aeb6-4775-b905-68918b09c5e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTMxNjc4MzUsIm5iZiI6MTc1MzE2NzUzNSwicGF0aCI6Ii85NjEyODcyNC80NjkwMTkwODgtNTlmNzYwMDUtYWViNi00Nzc1LWI5MDUtNjg5MThiMDljNWUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzIyVDA2NTg1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1OGY2NzRlZDk3NjQ4NDBmMzE0MDRjNjRhMjhlNmE4YWJjOGFhYTJiMGQyYTUzNWViOTZkYWEwMTg3MmJhMGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Eehjh7gCXDUVr7aLGjbq3yvWWTUQyUb2OJMStm5i8Ps">
> Regards, Andy

At first ,i set pre-define tag told by readme.md :`-DARM_TOOLCHAIN_PATH="C:/LLVMEmbeddedToolchainForArm-17.0.1-Windows-x86_64/bin" -DCMAKE_TOOLCHAIN_FILE=cmake/llvm.cmake` ,but it seems didnt work. After a long try, i found it need to set environment varies like this:

[image]

closed 2025-07-22 by Mr-P-hainan